Luro02/klipper-mpc #12: Installation issues

wildBill83, 2024-01-08:
Currently having problems installing on a Raspberry Pi 3B Klipper host. 
1. The initial installation instructions use the wrong command to clone the repository, it should be:
git clone https://github.com/Luro02/klipper-mpc
not
git clone git@github.com:Luro02/klipper-mpc.git
This avoids auth errors.
2. The instructions leave out "chmod +x ./install.sh" to make the script executable
3. I don't think this is compatible with the default version of Python on the Raspberry Pi, when executing ./install.sh I get the following error relating to Python, most likely requiring Python higher than 3.7.3:
Traceback (most recent call last):
  File "install.py", line 10, in <module>
    class Shell:
  File "install.py", line 33, in Shell
    def execute(self, command: list[str]):
TypeError: 'type' object is not subscriptable

chrisheib, 2024-02-29:
Ran into 1 and 2 aswell, didn't encounter 3.

What I did find was that https://github.com/Luro02/klipper-mpc/blob/9d4567c66827466402843e48a4e408b2e681f358/install.py#L50 didn't work on my py. Needed to comment it out. 

Also the integration into moonraker update doesn't seem to work for me.

chrisheib, 2024-02-29:
Removing `is_system_service: False` from the moonraker config shows klipper-mpc correctly in the update manager.