Luro02/klipper-mpc #14: Calibration Error

SHKinsem, 2024-01-29:
I was running the calibration gcode and immediately after it heated up, the error pops out:

```
error: the calculated target temperature '199.340356337887' must be greater than 200.00 and lower than 300.00.
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
```

Luro02, 2024-01-29:
Ahhh, interesting. Was your printer completely cold before starting calibration?

The idea is that it heats the printer from ambient temp to 100, then from 100 to 200 and measures how much it changes while doing that. After it reached 200°C it will do some calculations to find an estimate target temperature (i am not 100% sure what this temp is supposed to represent). In most circumstances it should be above 200°C and somewhere below your maximum temperature.

I remember that it failed for me when my printer wasn't cold and it assumed an ambient temperature of like 30°C, even though it was only 20°C room temperature.

You could try the other strategy, I think it is called differential tuning?

But keep in mind that the code is alpha and I haven't been able to get the last stage working correctly (the part cooling fan calibration)

SHKinsem, 2024-01-29:
> Ahhh, interesting. Was your printer completely cold before starting calibration?
>
> The idea is that it heats the printer from ambient temp to 100, then from 100 to 200 and measures how much it [...]

After waiting for it to become completely cold, it still isn't working. I think the main reason is that my hotend setup is highly responsive to temperature, eg. a Bambulab hotend clone and a 70W ceramic heater. This setup has a small heat capacity, and the thermometer is very close to the high-power heater.

[image]
From the graph, you can hardly find the peak of overshoot even if the heater was turned off from 100% power...

> But keep in mind that the code is alpha and I haven't been able to get the last stage working correctly (the part cooling fan calibration)

Wow, I thought this module could already be used for printing. Maybe you can put a tag to indicate that? I am very happy to see someone finally implementing this for Klipper. I have had a taste of MPC in my other machine running Marlin, and it is working flawlessly. The temperature fluctuation is so small and stable. Klipper's PID does a mess with my current hotend setup; the temperature fluctuates A LOT. That is the reason I was seeking for this.

SHKinsem, 2024-01-30:
[image]
Attached is the graph of my setup using pid temp control

Luro02, 2024-01-30:
https://klipper.discourse.group/t/model-predictive-temperature-control-port-for-klipper/12850